YvZheng/pycwr #12: CC雷达数据读取时三种异常

JiangXintongOUC, 2019-12-25:
在使用CC雷达数据进行测试时，出现三种类型异常。
一种是：
---> 55         assert BaseDataHeader_dict['ObsParam1']['ucScanMode'] > 100, "only vol support!"
     56         self.nsweeps = BaseDataHeader_dict['ObsParam1']['ucScanMode'] - 100
     57         BaseDataHeader_dict['CutConfig'] = np.frombuffer(buf_header, \
AssertionError: only vol support!

一种是：
     39         """检查雷达数据是否完整"""
     40         buf_radial_data = self.fid.read()
---> 41         assert len(buf_radial_data) == self.nrays * dtype_cc.PerRadialSize, "CC basedata size has problems!"
     42         return
     43 
AssertionError: CC basedata size has problems!

一种是：
     57         BaseDataHeader_dict['CutConfig'] = np.frombuffer(buf_header, \
     58                                                        dtype_cc.BaseDataHeader['CutConfigX30'],
                                                                    count=self.nsweeps,
---> 59                                                          offset=dtype_cc.CutSize_pos)
     60         ##解码第二部分观测参数
     61         BaseDataHeader_dict['ObsParam2'], _ = _unpack_from_buf(buf_header, \
ValueError: buffer is smaller than requested size

请问要如何解决？是否是数据源问题？

daxinnan, 2019-12-25:
CC雷达数据存在两种数据格式`CC1.0`和`CC2.0`。由于缺少基数据支持，目前只提供了`CC1.0`的读取方法，**请先确认CC雷达数据格式**。具体判别方法参见[新一代天气雷达基数据文件格式自动识别方法研究](http://gb.oversea.cnki.net/KCMS/detail/detail.aspx?filename=JYXH201307048&dbcode=CJFD&dbname=CJFD2013)

YvZheng, 2019-12-25:
@JiangXintongOUC 能否提供一下数据？

JiangXintongOUC, 2019-12-25:
您好，

文件里包含了三种出现异常的数据。
感谢！

YvZheng, 2019-12-25:
@JiangXintongOUC thx, and I will fix it soon.

JiangXintongOUC, 2020-03-18:
您好，我曾在Github上给您留言CC雷达数据读取异常问题，并给您发送了三个CC波段的雷达基数据，请问现在更新的pycwr版本能对这种波段的数据进行读取了吗？

@JiangXintongOUC

YvZheng, 2020-03-18:
> 您好，我曾在Github上给您留言CC雷达数据读取异常问题，并给您发送了三个CC波段的雷达基数据，请问现在更新的pycwr版本能对这种波段的数据进行读取了吗？
> 
> @JiangXintongOUC
> 
> 
> 
> 
> @JiangXintongOUC thx, and I will fix it soon.

我的邮箱没有收到cc的基数据，要不尝试重新发送到YuZheng1206@outlook.com这个邮箱

JiangXintongOUC, 2020-03-19:
已发送，附件三个基数据分别对应了三种异常，请您查收。

YvZheng, 2020-12-17:
#12 已修复

closed 2020-12-17 by YvZheng