YvZheng/pycwr #18: Import error with Cython support

zxdawn, 2020-03-15:
## Code

```
from pycwr.draw.RadarPlot import Graph, GraphMap
```

## Error
```
Traceback (most recent call last):
  File "crf.py", line 4, in <module>
    from pycwr.draw.RadarPlot import Graph, GraphMap
  File "/yin_raid/xin/miniconda3/envs/pycwr/lib/python3.7/site-packages/pycwr-0.2.14-py3.7-linux-x86_64.egg/pycwr/__init__.py", line 1, in <module>
    from . import configure, core, draw, io, interp
  File "/yin_raid/xin/miniconda3/envs/pycwr/lib/python3.7/site-packages/pycwr-0.2.14-py3.7-linux-x86_64.egg/pycwr/core/__init__.py", line 1, in <module>
    from . import NRadar, PyartRadar, transforms, RadarGrid, RadarGridC
  File "/yin_raid/xin/miniconda3/envs/pycwr/lib/python3.7/site-packages/pycwr-0.2.14-py3.7-linux-x86_64.egg/pycwr/core/NRadar.py", line 13, in <module>
    from .RadarGridC import get_CR_xy, get_CAPPI_xy
ImportError: /yin_raid/xin/miniconda3/envs/pycwr/lib/python3.7/site-packages/pycwr-0.2.14-py3.7-linux-x86_64.egg/pycwr/core/RadarGridC.cpython-37m-x86_64-linux-gnu.so: undefined symbol: __intel_sse2_strchr
```
## Version
0.2.14

## Solution
Specify compiler such as `gcc`:
```
CC=gcc pip install cython
CC=gcc python setup.py install
```
Or `icc`:
```
CC=icc pip install cython
LDSHARED="icc -shared" CC=icc python setup.py install
```

## Reference
[Can cython be compiled with icc?](https://stackoverflow.com/questions/37904377/can-cython-be-compiled-with-icc)

YvZheng, 2020-03-18:
最近忙着毕业答辩，毕业答辩结束修复该问题

YvZheng, 2020-12-17:
暂时不支持icc进行编译，windows平台有编译好的whl文件，请使用pip安装，Linux推荐使用gcc编译器

closed 2020-12-17 by YvZheng