YvZheng/pycwr #26: 无法读图

1209805090, 2020-08-14:
请问，这种问题是怎么回事儿？。。。
代码：
from pycwr.io import read_auto
from pycwr.draw.SingleRadarPlotMap import RadarGraphMap
import matplotlib.pyplot as plt

filename = r"F:/Z_RADR_I_Z9250_20180609220600_O_DOR_SA_CAP.bin.bz2"

PRD = read_auto(filename)
graph = RadarGraphMap(PRD)
graph.plot(0, "dBZ", continuously=True)
plt.show()

报错：
E:\Program Files\JetBrains\PyCharm 2019.1\helpers\pycharm_matplotlib_backend\backend_interagg.py:64: UserWarning: This figure includes Axes that are not compatible with tight_layout, so results might be incorrect.
  self.figure.tight_layout()
E:\Software\anaconda3\envs\snowflake\lib\site-packages\cartopy\io\__init__.py:260: DownloadWarning: Downloading: https://naciscdn.org/naturalearth/50m/physical/ne_50m_ocean.zip
  warnings.warn('Downloading: {}'.format(url), DownloadWarning)
Traceback (most recent call last):
  File "F:/源码/PycharmProjects/pythonLearning/read.py", line 10, in <module>
    plt.show()
  File "E:\Software\anaconda3\envs\snowflake\lib\site-packages\matplotlib\pyplot.py", line 272, in show
    return _show(*args, **kw)

bugsuse, 2020-08-16:
请给出你所使用的版本号以及相关的环境信息，比如pycwr，matplotlib的版本。

1209805090, 2020-08-19:
Package          Version
---------------- -------------------
appdirs          1.4.4
biopython        1.77
brotlipy         0.7.0
Cartopy          0.18.0
certifi          2020.6.20
cffi             1.14.0
cftime           1.2.1
chardet          3.0.4
cinrad           1.6.1
click            7.1.2
cryptography     2.9.2
cycler           0.10.0
Cython           0.29.21
decorator        4.4.2
easydict         1.9
Flask            1.1.2
geos             0.2.2
idna             2.10
ipython-genutils 0.2.0
itsdangerous     1.1.0
Jinja2           2.11.2
kiwisolver       1.2.0
lxml             4.5.2
MarkupSafe       1.1.1
matplotlib       3.2.2
MetPy            0.12.2
mkl-fft          1.1.0
mkl-random       1.1.1
mkl-service      2.3.0
netCDF4          1.5.4
numpy            1.19.1
olefile          0.46
opencv-python    4.3.0.38
OWSLib           0.20.0
packaging        20.4
pandas           1.1.0
Pillow           7.2.0
Pint             0.14
pip              20.2.1
pooch            1.1.1
pycparser        2.20
pycwr            0.2.15
pyepsg           0.4.0
pykdtree         1.3.1
pyOpenSSL        19.1.0
pyparsing        2.4.7
pyproj           2.6.1.post1
PyQt5            5.12
PyQt5-sip        4.19.19
PyQtWebEngine    5.12
pyshp            2.1.0
PySocks          1.7.1
python-dateutil  2.8.1
pytz             2020.1
PyYAML           5.3.1
requests         2.24.0
scipy            1.5.0
setuptools       49.2.1.post20200807
Shapely          1.7.0
sip              4.19.13
six              1.15.0
torch            1.6.0
torchvision      0.7.0
tornado          6.0.4
traitlets        4.3.3
urllib3          1.25.10
vanadis          0.0.2
Werkzeug         1.0.1
wheel            0.34.2
win-inet-pton    1.1.0
wincertstore     0.2
xarray           0.16.0

1209805090, 2020-08-19:
通过图像界面可以将图读出，但test文件夹中的读图代码确报错。。。

1209805090, 2020-08-19:
多谢，问题已解决。原因：cartopy运行程序默认从Natural Earth下载地图文件。可以通过下载地图文件放在本地文件夹下解决。

closed 2020-08-20 by bugsuse