YvZheng/pycwr #30: 请教一下，如何计算生成的图片的经纬度位置。

cmaMrSober, 2021-04-19:
老师您好，请问透明图片生成后，如何计算出左上角和右下角的经纬度，以便于叠加至WEBGIS（例如百度地图）上面。

YvZheng, 2021-04-21:
请问您是用什么函数生成的网格

cmaMrSober, 2021-04-23:
PRD = read_auto(r"D:\Z_RADR_I_Z9094_20210420155200_O_DOR_CAD_CAP_FMT.bin.bz2")
fig=plt.figure()
ax = plt.axes(projection=ccrs.PlateCarree())
graph = Graph(PRD)
graph.plot_crf(ax, cmap="pyart_NWSRef",cbar=False)
ax.spines['right'].set_visible(False)
ax.spines['top'].set_visible(False)
ax.spines['left'].set_visible(False)
ax.spines['bottom'].set_visible(False)
ax.set_xticks([])
ax.set_yticks([])
# plt.show()
fig.savefig(r"D:\1.png",dpi=600)

YvZheng, 2021-04-26:
您还是直接导出组合反射率数据吧，用PRD.add_product_CR_lonlat(XLon, YLat)函数来生成组合反射率的网格，在PRD.product下面可以查看，生成的数据是xarray的数组，附带经纬度信息，然后利用imshow绘图即可，然后可以叠加在地图上。

cmaMrSober, 2021-04-26:
好的，谢谢老师（学弟？）^.^ hhhhh

cmaMrSober, 2021-09-10:
> #结题
`
#生成透明图片，并确定图片经纬度位置。
```
def graph(fname):
    PRD = read_auto(fname)
    print(PRD.scan_info)
    lat=PRD.fields[0].lat.values
    lon=PRD.fields[0].lon.values
    extent=[lon.min(), lon.max(), lat.min(), lat.max()]                    
    ex= [ str(x) for x in extent]
    tr ="_GIS_"+'_'.join(ex)
    print(tr)
    proj = ccrs.AzimuthalEquidistant(
                    central_longitude=PRD.scan_info.longitude,
                    central_latitude=PRD.scan_info.latitude,
                )
    fig=plt.figure(figsize=(10, 10), dpi=160)
    ax = fig.add_axes([0,0,1,1], projection=proj)
    graph = Graph(PRD)
    graph.plot_crf(ax, cmap=“pyart_NWSRef”,cbar=False)
    plt.axis("off")
    plt.margins(0,0)
    filenameWithoutRex=fname[0:fname.find('.')]
    imgFileWithPath=filenameWithoutRex+tr+"_CR.png"
    fig.savefig(imgFileWithPath,transparent=True, pad_inches=0)
    plt.close("all")
```
`

closed 2024-05-13 by YvZheng