YvZheng/pycwr #39: 径向扫描范围问题

jiaobf, 2022-04-12:
[image]

左图是我用pycwr解出来的雷达数据，放进arps的结果，右图是其他程序解码的结构。pycwr径向扫描范围好像不够，是否设置了扫描范围上限呢？

hepengz, 2022-04-13:
之前对于WSR98D数据遇到过类似的问题，但是不知道你这个雷达数据是什么格式，

对于我们国家业务的双偏振雷达数据，0.5和1.5°仰角 径向速度V和强度Z的PRF是不一样的，探测距离有差异，即数据的距离库有差异，V-1320-330km[250m分辨率] 而Z-1840-460km

pycwr径向扫描范围的最大距离库，对于WSR98D格式雷达来说，默认使用的是所有仰角V的最大距离库数：1320

所以要解析绘制底层超过330km之外的回波数据[一般超过330km，回波高度已经达到8km附近了，所以pycwr默认的1320最大库影响也不大]，可以修改pycwr.io.WSR98DFile.py文件中：
https://github.com/YvZheng/pycwr/blob/8013dd6c597618949c5fcbf86e38502525a8136d/pycwr/io/WSR98DFile.py#L346

return np.array([self.radial[idx]['fields']['V'].size for idx in self.sweep_start_ray_index])
-> return np.array([self.radial[idx]['fields']['dBZ'].size for idx in self.sweep_start_ray_index])

或者更简单一点，直接在https://github.com/YvZheng/pycwr/blob/8013dd6c597618949c5fcbf86e38502525a8136d/pycwr/io/WSR98DFile.py#L217
设置self.max_bins = 1840 [想要的距离库数目]

如果你的数据不是WSR98D，可以类比试一下

jiaobf, 2022-04-13:
> 之前对于WSR98D数据遇到过类似的问题，但是不知道你这个雷达数据是什么格式，
> 
> 对于我们国家业务的双偏振雷达数据，0.5和1.5°仰角 径向速度V和强度Z的PRF是不一样的，探测距离有差异，即数据的距离库有差异，V-1320-330km[250m分辨率] 而Z-1840-460km
> 
> pycwr径向扫描范围的最大距离库，对于WSR98D格式雷达来说，默认使用的是所有仰角V的最大距离库数：1320
> 
> 所以要解析绘制底层超过330km之外的回波数据[一般超过330km，回波高度已经达到8km附近了，所以pycwr默认的1320最大库影响也不大]，可以修改pycwr.io.WSR98DFile.py文件中：
> 
> https://github.com/YvZheng/pycwr/blob/8013dd6c597618949c5fcbf86e38502525a8136d/pycwr/io/WSR98DFile.py#L346
> 
> return np.array([self.radial[idx]['fields']['V'].size for idx in self.sweep_start_ray_index]) -> return np.array([self.radial[idx]['fields']['dBZ'].size for idx in self.sweep_start_ray_index])
> 
> 或者更简单一点(这个方法适用于pyart-Radar对象，因为pycwr-PRD对象需要用到bins_per_sweep，这个方法不行，需要修改bins_per_sweep)，直接在
> 
> https://github.com/YvZheng/pycwr/blob/8013dd6c597618949c5fcbf86e38502525a8136d/pycwr/io/WSR98DFile.py#L217
> 
> 
> 设置self.max_bins = 1840 [想要的距离库数目]
> 如果你的数据不是WSR98D，可以类比试一下

感谢您的回答，这个是WSR98D数据，通过修改
return np.array([self.radial[idx]['fields']['V'].size for idx in self.sweep_start_ray_index])
-> return np.array([self.radial[idx]['fields']['dBZ'].size for idx in self.sweep_start_ray_index])
问题解决了。我看了下，径向格点数最多能有2600+。

感谢~

closed 2022-04-13 by jiaobf