YvZheng/pycwr #46: 关于安装的问题

FOR-BANANA, 2023-07-21:
安装完后，在导入时问题报错：“ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()”
[image]

FOR-BANANA, 2023-07-21:
请问这个要如何解决？

gf1023456, 2023-08-15:
同遇到，解决了吗

gf1023456, 2023-08-15:
> 

pycwr 包中的 cm_colorblind.py 文件的第 63 行。有误 你将那个if改为   if np.any([x == 'red' for x in spec]):  这样就能用了

YvZheng, 2023-08-23:
I installed pycwr 3.6.0 based on python3.11 and found no errors. Can you provide me with specific environment information? @FOR-BANANA @gf1023456 
[image]

Qxaqq, 2023-09-28:
请帮忙指导一下
[image]

gf1023456, 2023-09-28:
这是来自QQ邮箱的假期自动回复邮件。 您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

Qxaqq, 2023-09-28:
> 请帮忙指导一下 [image]

已解决，按照Pull requests里的修改，3处if 'red' in spec:改为 if np.any([x == 'red' for x in spec]):

Syize, 2024-05-06:
> I installed pycwr 0.3.6 based on python3.11 and found no errors. Can you provide me with specific environment information? @FOR-BANANA @gf1023456 ![image](https://private-user-images.githubusercontent.com/34502899/262611824-baf6c790-ed16-4bca-b960-607ffc4ce70d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ5OTg5OTcsIm5iZiI6MTcxNDk5ODY5NywicGF0aCI6Ii8zNDUwMjg5OS8yNjI2MTE4MjQtYmFmNmM3OTAtZWQxNi00YmNhLWI5NjAtNjA3ZmZjNGNlNzBkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTA2VDEyMzEzN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTIyNmUwZTQ4OTE0NDk3MjI3MTYyOTAwNjRlNzViOGEwNDZiOTkxZGY2MTRhYjUxYjZiMjM1YzVlMjQwMmQ2OTAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.pSHH1_qbRYqFa2dIGRi_4iC7-UbkA8frlIyKC861f20)

This error happens because the type of variable `spec` changes several times. By adding `print(spec)` we can see the type of `spec` can be:

```
<class 'dict'>
<class 'numpy.ndarray'>
<class 'list'>
```
and the error raises when the type of `spec` is `list`.

YvZheng, 2024-05-13:
This problem has been fixed in version 0.4.0

closed 2024-05-13 by YvZheng