YvZheng/pycwr #50: Import pycwr时报错

December0w0, 2024-01-26:
在使用pycwr的时候，其会在import时报错。报错信息如图所示。我的python版本是3.10.9，numpy版本1.26.3。请问这种情况应该怎么处理？
[image]

jshn9515, 2024-02-05:
打开cm_colorblind.py文件，将以下代码：
```python
with warnings.catch_warnings():
    warnings.simplefilter("ignore", FutureWarning)
    if 'red' in spec:
        return colors.LinearSegmentedColormap(name, spec, lutsize)
    else:
        return colors.LinearSegmentedColormap.from_list(name, spec, lutsize)
```
替换为：
```python
with warnings.catch_warnings():
    warnings.simplefilter("ignore", FutureWarning)
    if isinstance(spec, dict) and "red" in spec.keys():
        return colors.LinearSegmentedColormap(name, spec, lutsize)
    else:
        return colors.LinearSegmentedColormap.from_list(name, spec, lutsize)
```

December0w0, 2024-02-20:
> 打开cm_colorblind.py文件，将以下代码：
>
> ```python
> with warnings.catch_warnings():
>     warnings.simplefilter("ignore", FutureWarning)
>     if 'red' in spec:
>         return colors.LinearSegmentedColormap(name, [...]

修改完成了，还是有问题。但是后面排查之后，发现是这个包和wrf-python有冲突。已更换安装环境，问题解决。

closed 2024-02-20 by December0w0